armel/uv-k5-firmware-custom #401: Changing SetNFM causes radio interference

DanielGekeler, 2025-05-17:
When changing SetNFM in the menu and exiting the setting, a pulse of radiation is emitted. I noticed this as a blop sound on my Speakers. This happens despite F LOCK being set to DISABLE ALL. The Amplitude of the interference does correlate with the selected power setting. Unfortunately I do not have any radio equipment to investigate this further.

This could be an Issue with the hardware itself or a bug in the software.
I did not test it with other firmwares.

# Hardware
Regular Quansheng UV-K5

# Software
F4HWN Bandscope v4.0

# Steps to reproduce
1. Go to a frequency like 433.100MHz in FM
2. Set a high power level to make the interference easier to detect
3. Go into the Menu and select SetNFM (Option 68)
4. Press the Menu button to select it
5. (Optional) pick another value
6. Press the Menu button to apply the change
7. A short pulse of radiation is emitted

armel, 2025-05-17:
Hello Daniel,

The reason is simple: you've most likely activated the [VOX](https://github.com/armel/uv-k5-firmware-custom/wiki/Button-functions#7-vox) function. You can see it in the status bar by the VX indicator. Make sure menu 53/68 is set to OFF — or, more easily, press and hold the 7 key to disable it.

When VOX is enabled, the radio may start transmitting automatically, sometimes just from ambient noise or even **a simple key press**. And to be honest, the VOX on the K5 isn’t particularly reliable — no big surprise for a $20 device.

Armel, F4HWN

DanielGekeler, 2025-05-17:
Thank you for looking into this.
I just reproduced the issue and confirmed that VOX is disabled. The issue persists.

Would you like to see a Video?

armel, 2025-05-17:
Sure, a video could definitely help to better understand the issue.

Armel, F4HWN

DanielGekeler, 2025-05-17:
Here is a video of me applying the setting close to a speaker. The noise is audible.
I can assure you it is emitted by the speaker. Doing it further away from the speaker lowers the volume.
It happens in the exact moment I press the Menu button.

https://github.com/user-attachments/assets/bc40f563-fb14-4782-af5f-42fbecf2c42d

DanielGekeler, 2025-05-17:
It's strange that this only happens when changing this particular setting.
Increasing the power output makes the noise louder.

I assume it's a glitch in the chip that handles the radio part.
We are going to have to live with it.

armel, 2025-05-17:
> It's strange that this only happens when changing this particular setting.
> Increasing the power output makes the noise louder.

Yes and no, it’s not really surprising. This menu modifies settings at the BK4819 level, which directly triggers GPIO access and other internal operations. That’s not the case with most other settings, which tend to be more passive and only change some values directly in memory. 

Also, we can’t expect too much from a $20 radio: there’s barely any shielding or filtering. The whole thing radiates noise — the MCU clock, the PWM driving the screen backlight... everything leaks.

Last but not least, let’s be honest : you’re dealing with a rather unusual use case. Not many people operate this kind of radio right next to a JBL speaker... 😉

Armel, F4HWN

closed 2025-05-18 by DanielGekeler