armel/uv-k5-firmware-custom #412: Spectrum analyzer forget settings, after switch off

P1-code, 2025-06-28:
I set these parameters to listen: [https://ibb.co/0yb6kx1k](url)
After switch off/switch on I got these parameters: [https://ibb.co/Fb91b1dd](url)
Forgets many settings( frequency step, bandwidth,...). I think it's a bug.

The radio is running the 4.0 "Bandscope" edition.

armel, 2025-06-28:
Hello,

Your links don't seem to be working. So, I’m not sure which parameters you're referring to.

Please note that only three parameters are currently saved:
- settings.scanStepIndex
- settings.stepsCount
- settings.listenBw

That’s all for now.
At the moment, I don’t have enough EEPROM space to store additional settings.

Armel, F4HWN

closed 2025-06-28 by P1-code

armel, 2025-06-28:
Ok, now I can see the pictures. Tahnk you 👍 
The highlighted parameter is one of those that can be saved. However, be aware that the save only happens when exiting the bandscope properly by pressing **EXIT**.

Armel, F4HWN