armel/uv-k5-firmware-custom #459: RX on top of ScnRng

bartwas78-max, 2025-08-13:
When setting scan range and signal is received by VFO occupying ScnRng line, the RX letters appear on top of ScnRng

armel, 2025-08-13:
I’m not sure I fully understand the issue yet.
Could you please be more specific or share a short photo/video showing the problem?

It would help if you include:

- Firmware version and edition (e.g., F4HWN vX.Y – Bandscope/Broadcast/RescueOps/Basic)
- Exact steps to reproduce (menus you open, ScnRng value, how you start the scan, VFO A or B)
- What you expected vs. what you see (where exactly the “RX” letters overlap the ScnRng line)

Armel, F4HWN

bartwas78-max, 2025-08-13:
By the way, excellent firmware. You really made this radio sing and dance

bartwas78-max, 2025-08-13:
Was setting bandscan range and noticed this issue, not a big deal, but it dragged my attention

bartwas78-max, 2025-08-13:
It was bandscope v4.1, just upgraded to bandscope 4.2 and it does the same

armel, 2025-08-13:
Ok thank you. Well, I tried to reproduce the bug, but no luck. I tested several approaches, but so far I can’t get the overlap to occur. At the beginning of the video, I followed your instructions as described:

- Set VFO B to 145.800, 
- Switch to VFO A and set it to 145.200,
- Hold button 5 to display ScnRng 145.200/145.800. 

And use other radio to transmit on 145.800...

https://github.com/user-attachments/assets/625c1cf8-2fc9-4c38-9b8e-e6e2ef11d6ea

armel, 2025-08-13:
That’s why I set ScnRng to the top from 1:00 onward and retried.

bartwas78-max, 2025-08-13:
Mist have been something with my radio, I did full reset and it is working fine now.
Sorry to waste your time :(

armel, 2025-08-13:
No worries at all and thanks for the update.
I’m glad a full reset fixed it. 
You definitely didn’t waste my time. Reports like yours help me to improve things. If it happens again, please note your firmware version/edition and the steps (a short video helps), and we’ll dig in.

Regards,

Armel, F4HWN

closed 2025-08-13 by armel