armel/uv-k5-firmware-custom #467: Feature request: PTT function for F1 / F2 long press options

geckolinux, 2025-09-06:
Hi there, first of all thanks very much for maintaining this featureful firmware!

There is a feature that I really like in https://github.com/losehu/uv-k5-firmware-custom that could possibly be implemented in F4HWN. It allows for making the F1 and/or F2 keys work as PTT keys for the A or B frequency. I'm not a very good coder, but this is how it's implemented:

- The English menu option is defined here: https://github.com/losehu/uv-k5-firmware-custom/blob/main/chinese.h
     - `#define 主信道发射 "MAIN SEND"` (This makes the key transmit the A frequency)
     - `#define 副信道发射 "DUAL SEND"`(This makes the key transmit the B frequency)
- A search for `主信道发射` and `副信道发射` finds several references in https://github.com/losehu/uv-k5-firmware-custom/blob/main/ui/menu.c
```
#ifdef ENABLE_SIDEFUNCTIONS_SEND
               {主信道发射, ACTION_OPT_SEND_CURRENT},
               {副信道发射, ACTION_OPT_SEND_OTHER},
#endif 
```
- Those functions are defined or used in https://github.com/losehu/uv-k5-firmware-custom/blob/main/app/action.c and https://github.com/losehu/uv-k5-firmware-custom/blob/main/settings.h

Frost-Smart, 2025-09-06:
Не считаю что это остро необходимо ! 
Скорее всего вообще не нужно ... 
( использование F1 и F2 в качестве РТТ)

geckolinux, 2025-09-06:
@Frost-Smart That's why it would be available as an option, and not enabled by default. Those who don't need it won't enable it.

armel, 2025-09-07:
Hello,

Thanks for the suggestion and the reference to Losehu’s firmware.

I get the idea of turning F1/F2 into PTT A/B, but I try to keep the UX simple and consistent. Right now, a long-press on 2 key already toggles A↔B very quickly, then PTT works as expected on the active VFO.

Concerns with mapping F1/F2 to PTT:

- Accidental TX risk (pressing a function key that became PTT).
- Shortcut conflicts and loss of consistency for F1/F2.
- Extra UI/code complexity for a small benefit, while the current flow is already fast (long-press on 2 key → PTT).
- Flash memory constraints...

If this gets broader demand, I could consider an another build flag, but I’d rather avoid UI bloat for now. In the meantime, long-press on 2 key remains the simplest and most reliable way to pick A or B before transmitting.

73

Armel, F4HWN